TheRobotStudio/SO-ARM100 #11: Open CAD files format

kolergy, 2024-10-21:
First thanks for sharing this design and including the CAD files!

However these are in the proprietary Solidworks format (.SDLPRT). Unfortunately most people do not have or can not afford Solidworks.

Would it be possible to share the CAD files in an open CAD format like STEP?

TheRobotStudio, 2024-10-26:
Yes, I will upload the STEP on Monday.

brukg, 2024-12-08:
@TheRobotStudio  It would be great if we can have the step file for each part instead of the whole model. is that possible to do?

pkooij, 2025-01-27:
I added the separate cad files

closed 2025-01-27 by pkooij

odellus, 2025-03-30:
@pkooij how big of a lift is it to ask to convert the [7 DOF SLDASM](https://github.com/TheRobotStudio/SO-ARM100/tree/main/URDF/SO_7DOF_ARM100_08h_HL_01d_URDF_01.SLDASM) model to STEP format? I'm terribly interested in the 7 DOF version and I just looked up price of solidworks so yeah. 🙏🏻

pkooij, 2025-03-31:
Hi @odellus,

That 7DOF file is an early version of HopeJr arm (I will remove it from this repo), that can now be found in this repo: https://github.com/TheRobotStudio/HOPEJr