TheRobotStudio/SO-ARM100 #12: Base Idle Clap tolerances too tight

brainwavecoder9, 2024-10-26:
Probably overkill to add these as issues, and I would consider this VERY minor, but realizing that might be best way to track.

The tolerance on the base idle clamp (pictured) is too tight. When using the 3rd horn on servo 1, the clamp interferes with the horn.

Wouldn't matter if the clamp side of the horn was intended not to move, and I'm sure the servo can overcome it, but the horn is screwed onto the arm so this will cause friction.

[image]

TheRobotStudio, 2024-10-26:
So when you screw the clamp on it grips the extra servo horn? It should be slightly loose, just a 0.1mm clearance. If the clamp is not fully screwed down does that allow some movement? If needed I can add a bit more clearance...

brainwavecoder9, 2024-10-26:
Yes it grips tightly. I was able to carve it out to make clearance.

It's also possible it has to do with my print and/or my material possibly warping. Someone talked me into PC (do not recommend). ABS and PETG have been very good.

brainwavecoder9, 2024-10-26:
FYI Confirmed the interference with ABS also