TheRobotStudio/SO-ARM100 #13: Adding SO-ARM100 to Mujoco Menagerie

chernyadev, 2024-10-28:
Thank you for this excellent project!

I’ve been working on integrating the SO-ARM100 v1.3 into Mujoco. You can find the pull request adding your robot to the Mujoco Menagerie repository here: [PR #105 on Mujoco Menagerie](https://github.com/google-deepmind/mujoco_menagerie/pull/105).

To complete the PR, a license file is required. I didn’t find a license file in this repository, though from reviewing the URDF files, it appears you might be using the BSD license.

Could you please confirm the license and, if possible, provide a license file for inclusion with this PR?

Thank you in advance for your assistance!

TheRobotStudio, 2024-10-28:
Thanks. I've added the Apache 2 licence in the main - will that allow you to complete the PR?

chernyadev, 2024-10-28:
Yes, absolutely, thank you very much!

closed 2024-10-28 by chernyadev