TheRobotStudio/SO-ARM100 #15: parts list quantity question

MatthewNinety, 2024-10-31:
In your parts list, it says the unit price for servos is $14.  But when I click the US alibaba link that you have listed for the servos, it says there are 6 servos per batch, and the total price is about $17 per batch, with about $12 in shipping.
That means the total price for 12 servos (17*2)  and shipping is about $50, which comes out to about $2 per servo.
This is a significant difference.  Did I make a mistake?

TheRobotStudio, 2024-11-01:
If you follow it through that 1 piece should add 6 servos to the cart and an adaptor board, now 15USD each plus tax of about 17USD.

To France where I am I get free shipping presently so the total is 103EUR.

MatthewNinety, 2024-11-09:
Thanks that solved it! I act like I've never gone shopping before :)

closed 2024-11-09 by MatthewNinety