TheRobotStudio/SO-ARM100 #16: Are these links for China Mainland the same parts as in the list?

Vacuame, 2024-11-05:
STS3215 Servo：https://item.taobao.com/item.htm?id=769061524145
Motor Control Board：https://item.taobao.com/item.htm?id=738955630278

I found these two part in chinese shop, but uncertainty about feasibility

TheRobotStudio, 2024-11-05:
They look correct - are you able to use the links on Ali provided in the readme?

Vacuame, 2024-11-06:
> They look correct - are you able to use the links on Ali provided in the readme?

I haven't tried it yet, because it says overseas buyers only. Maybe I can buy after I sign up a overseas buyer, but still too hassle.

TheRobotStudio, 2024-11-08:
That seems odd to me, I am checking with the company. You can also get in touch directly with Jenny Deng who is the sales rep that runs the Ali page: 2337371583@qq.com

I'd be interested to hear how you get on.

TheRobotStudio, 2024-11-08:
Feetech confirm that you can buy through the Taobao store for delivery to China.

Vacuame, 2024-11-26:
> Feetech confirm that you can buy through the Taobao store for delivery to China.

We have fully tested the devices in the link and confirmed that they are available, so you can now add them to the Sourcing Parts list!