TheRobotStudio/SO-ARM100 #24: Loose Base and Shoulder Joints Causing Vibration in Robotic Arm – Possible Solutions?

nomorewzx, 2024-11-29:
Hi, @TheRobotStudio thanks a lot for your excellent work! 

After assembling my robotic arm, I noticed that the Base Rotation Joint and Shoulder Joint of followe arm are a bit loose. 

Even after tightening all the screws, there’s still some play in these joints. This causes the entire arm to vibrate during teleoperation. 

Could this issue possibly be caused by insufficient precision in the 3D-printed links? 

Or Has anyone else encountered this issue? How did you solve it?

P.S. I am using the FeeTechSTS3215 7.4V version motors

Thanks a lot!

TheRobotStudio, 2024-11-29:
Some people have reported problems with the 12v version but I have not encountered it myself yet so I'm not clear on the causes.

Where is the looseness? Where the structure screws to the servo horns or how the servos attach to the arm itself?

nomorewzx, 2024-12-03:
@TheRobotStudio you are correct that the looseness in my video is actually the normal thing.

It turns out that the vibration is not due to any loose connection but due to the motor configuration.

I changed the minimum start force of rotation base motor and shoulder motor to 5N (16N by default) and reduced the P_coefficient to 8 (16 by default), then the vibration is almost gone.

TheRobotStudio, 2024-12-04:
Great!

closed 2024-12-04 by TheRobotStudio