TheRobotStudio/SO-ARM100 #27: If I'm making 2 arms, is there any reason not to by twelve of the "1 arm" servos?

jbrownkramer, 2025-01-01:
They're cheaper, and appear to be the same thing.

TheRobotStudio, 2025-01-01:
What do you mean by the 1 arm servos?

jbrownkramer, 2025-01-02:
In the read me, the parts list for two arms links to some STS3215 Servos that are $15 each, and the parts list for 1 arm is a different link to STS3215 Servos that are less than $14 each.

jbrownkramer, 2025-01-27:
Just to bump this again.  There are two different US links to the STS3215 Servos in README.md.  The first one is $15/each and the second one is < $14/each.  I ordered from the first link because it was in the part list for the building two arms, which is what I am doing.  The $15 version comes in lots of 6 with a controller board for each lot.  So when you buy 12 servos you're really buying the servos at about $14 each and about $12 for two controller boards.  But those boards are no longer used in the project.  I think the first link should be changed to point to the cheaper servos.