TheRobotStudio/SO-ARM100 #28: I think the parts list has redundant parts

jbrownkramer, 2025-01-18:
Below is the first parts list on on the README.  I started by going down the list and buying all of the things I didn't already have.  However, the first link includes 2 URT-1 boards, which I believe serve the same purpose as the Motor Control Boards in the second link.  At any rate, all of the videos I watched use the URT-1 board, and not the other Motor Control Boards.  The URT-1 boards come with USB cables and the ports are not USB C, making the third link unnecessary.  The URT-1s don't have a barrel jack power supply, making the fourth link unnecessary.  

It would be nice to list the requirement of a six-volt power supply with leads, since that's something I don't have.

If the linked motor-controller boards are meant as alternatives, it would be nice to have some instructions for installing drivers, which I haven't been able to find.

#### Parts For Two Arms (Standard Teleoperation Setup):
| Part | Amount | Unit Cost (US) | Buy US | Unit Cost (EU) | Buy EU |
|---|---|---|---|---|---|
| STS3215 Servo<sup>[1](#myfootnote1)</sup> | 12 | $15 | [Alibaba](https://www.alibaba.com/product-detail/6PCS-7-4V-STS3215-Servos-for_1600523509006.html) | 13€ | [Alibaba](https://www.alibaba.com/product-detail/6PCS-7-4V-STS3215-Servos-for_1600523509006.html) |
| Motor Control Board | 2 | $11 | [Amazon](https://www.amazon.com/Waveshare-Integrates-Control-Circuit-Supports/dp/B0CTMM4LWK/) | 12€ | [Amazon](https://www.amazon.fr/-/en/dp/B0CJ6TP3TP/)|
| USB-C Cable 2 pcs | 1 | $7 | [Amazon](https://www.amazon.com/Charging-etguuds-Charger-Braided-Compatible/dp/B0B8NWLLW2/?th=1) | 7€ | [Amazon](https://www.amazon.fr/dp/B07BNF842T/) |
| Power Supply<sup>[2](#myfootnote2)</sup> | 2 | $10 | [Amazon](https://www.amazon.com/Facmogu-Switching-Transformer-Compatible-5-5x2-1mm/dp/B087LY41PV/) | 13€ | [Amazon](https://www.amazon.fr/-/en/dp/B01HRR9GY4/) |
| Table Clamp 4pcs | 1 | $18 | [Amazon](https://www.amazon.com/WORKPRO-Clamps-Woodworking-One-Handed-Spreader/dp/B0CQYDJWS8/) | 21€ | [Amazon](https://www.amazon.fr/dp/B08HW9VFM8/)|
| Screwdriver Set<sup>[3](#myfootnote3)</sup> | 1 | $6 | [Amazon](https://www.amazon.com/Precision-Phillips-Screwdriver-Electronics-Computer/dp/B0DB227RTH) | 10€ | [Amazon](https://www.amazon.fr/dp/B08ZXVMVYD/) |
| Total |---| $241 |---| 244€ |---|

pkooij, 2025-01-27:
Hi, @jbrownkramer the motor control board is needed because it's supported in LeRobot [https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md](url). It's indeed a bit confusing that the first link comes with the URT-1 boards but they are not supported there.

jbrownkramer, 2025-01-27:
Hi, @pkooij that makes sense!  In that case I would suggest changing the list in the "Parts For Two Arms" section of the README.md to match the links in the "Parts For One Arm" section, since the user can save $12 and not buy any unnecessary parts.