TheRobotStudio/SO-ARM100 #31: Disaster: Built following the instructions here.  Only then started following Huggingface, which says I need to modify all leader servos.

jbrownkramer, 2025-01-27:
First question: Is it strictly necessary for me to modify all leader servos?  On of your videos mentions something about setting the servos to be powered.  Could setting them to be not powered in the case of the leader help here?

Second: Rather than linking to the assembly instructions for an older version of arm, would it be better to remove the link to the servo-reprogramming video and the assembly video, and instead link to the HuggingFace instructions for this arm?

https://github.com/huggingface/lerobot/blob/main/examples/10_use_so100.md

https://www.youtube.com/watch?v=FioA2oeFZ5I&ab_channel=HuggingFace

Cadene, 2025-01-27:
> "Is it strictly necessary for me to modify all leader servos?"
No it is not, but the leader arm will be harder to move due to friction. You wont be able to teleoperate it with one hand and for a long time. We are working on an updated version of the motors to not have to remove the gears.

> "link to the HuggingFace instructions for this arm"
Yes, really sorry for the trouble. The team is growing and we finally have the human ressources to iterate on the tutorial. We want to add checks to every assembly steps to avoid any issue.

jbrownkramer, 2025-01-27:
I really appreciate the response!  I have really been enjoying building these arms and I am looking forward to doing things with them.

closed 2025-02-07 by pkooij