TheRobotStudio/SO-ARM100 #54: Improved URDF

StoneT2000, 2025-04-03:
thanks for the great low-cost SO100 arm, been having a lot a fun playing with it recently.

I have interest in doing sim2real / simulation heavy experiments and found that the URDF in this repo is not well defined (missing joint limits / some joint axes are inverted, and lack of decomposed collision meshes)

I have a fixed version in my own github repo, happy to make a PR for it here. I saw some people having trouble importing it into their own sims on discord earlier.

shareshareneed, 2025-04-06:
> thanks for the great low-cost SO100 arm, been having a lot a fun playing with it recently.
>
> I have interest in doing sim2real / simulation heavy experiments and found that the URDF in this repo is [...]

Hello, may I ask which UPDF improvement project you are working on? I just started to contact this project recently, and I would like to refer to your improvement work

StoneT2000, 2025-04-06:
https://github.com/haosulab/ManiSkill/tree/main/mani_skill/assets/robots/so100 modified URDF here

just added so100 to my labs simulation framework for training/evaluation experiments and sim2real

pkooij, 2025-04-22:
Hi @StoneT2000, that's very nice would you like to make a PR?

StoneT2000, 2025-05-08:
Yes I'll make a PR this week.