TheRobotStudio/SO-ARM100 #60: GripperLeader calibration error

correderadiego, 2025-04-13:
I am having issues calibrating the leader board. I have managed to mount both arms but in the calibration process I am having the following error. I have tested with different servos in the gripper and the result is the same.

Do you have any idea? Thanks for all and for this amazing project. :)

python lerobot/scripts/control_robot.py --robot.type so100 --control.type teleoperate
INFO 2025-04-13 15:42:29 ol_robot.py:353 {'control': {'display_cameras': True, 'fps': None, 'teleop_time_s': None},
 'robot': {'calibration_dir': '.cache/calibration/so100',
           'cameras': {'laptop': {'camera_index': 0,
                                  'channels': 3,
                                  'color_mode': 'rgb',
                                  'fps': 30,
                                  'height': 480,
                                  'mock': False,
                                  'rotation': None,
                                  'width': 640},
                       'phone': {'camera_index': 2,
                                 'channels': 3,
                                 'color_mode': 'rgb',
                                 'fps': 30,
                                 'height': 480,
                                 'mock': False,
                                 'rotation': None,
                                 'width': 640}},
           'follower_arms': {'main': {'mock': False,
                                      'motors': {'elbow_flex': [3, 'sts3215'],
                                                 'gripper': [6, 'sts3215'],
                                                 'shoulder_lift': [2,
                                                                   'sts3215'],
                                                 'shoulder_pan': [1, 'sts3215'],
                                                 'wrist_flex': [4, 'sts3215'],
                                                 'wrist_roll': [5, 'sts3215']},
                                      'port': '/dev/ttyACM0'}},
           'gripper_open_degree': None,
           'leader_arms': {'main': {'mock': False,
                                    'motors': {'elbow_flex': [3, 'sts3215'],
                                               'gripper': [6, 'sts3215'],
                                               'shoulder_lift': [2, 'sts3215'],
                                               'shoulder_pan': [1, 'sts3215'],
                                               'wrist_flex': [4, 'sts3215'],
                                               'wrist_roll': [5, 'sts3215']},
                                    'port': '/dev/ttyACM1'}},
           'max_relative_target': None,
           'mock': False}}
Connecting main follower arm.
Connecting main leader arm.
Activating torque on main follower arm.
**Wrong motor position range detected for gripper. Expected to be in nominal range of [0, 100] % (a full linear translation), with a maximum range of [-10, 110] % to account for some imprecision during calibration, but present value is -127.23577117919922 %.** This might be due to a cable connection issue creating an artificial jump in motor values. You need to recalibrate by running: `python lerobot/scripts/control_robot.py calibrate`
Traceback (most recent call last):
  File "/home/ziash/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 397, in apply_calibration_autocorrect
    values = self.apply_calibration(values, motor_names)
  File "/home/ziash/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 467, in apply_calibration
    raise JointOutOfRangeError(
lerobot.common.robot_devices.motors.feetech.JointOutOfRangeError: Wrong motor position range detected for gripper. Expected to be in nominal range of [0, 100] % (a full linear translation), with a maximum range of [-10, 110] % to account for some imprecision during calibration, but present value is -127.23577117919922 %. This might be due to a cable connection issue creating an artificial jump in motor values. You need to recalibrate by running: `python lerobot/scripts/control_robot.py calibrate`

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ziash/lerobot/lerobot/scripts/control_robot.py", line 373, in <module>
    control_robot()
  File "/home/ziash/lerobot/lerobot/configs/parser.py", line 120, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/ziash/lerobot/lerobot/scripts/control_robot.py", line 360, in control_robot
    teleoperate(robot, cfg.control)
  File "/home/ziash/lerobot/lerobot/common/robot_devices/utils.py", line 28, in wrapper
    raise e
  File "/home/ziash/lerobot/lerobot/common/robot_devices/utils.py", line 24, in wrapper
    return func(robot, *args, **kwargs)
  File "/home/ziash/lerobot/lerobot/scripts/control_robot.py", line 205, in teleoperate
    control_loop(
  File "/home/ziash/lerobot/lerobot/common/datasets/image_writer.py", line 36, in wrapper
    raise e
  File "/home/ziash/lerobot/lerobot/common/datasets/image_writer.py", line 29, in wrapper
    return func(*args, **kwargs)
  File "/home/ziash/lerobot/lerobot/common/robot_devices/control_utils.py", line 216, in control_loop
    robot.connect()
  File "/home/ziash/lerobot/lerobot/common/robot_devices/robots/manipulator.py", line 272, in connect
    self.leader_arms[name].read("Present_Position")
  File "/home/ziash/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 757, in read
    values = self.apply_calibration_autocorrect(values, motor_names)
  File "/home/ziash/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 400, in apply_calibration_autocorrect
    self.autocorrect_calibration(values, motor_names)
  File "/home/ziash/lerobot/lerobot/common/robot_devices/motors/feetech.py", line 553, in autocorrect_calibration
    raise ValueError(f"No integer found between bounds [{low_factor=}, {upp_factor=}]")
ValueError: No integer found between bounds [low_factor=np.float32(0.15283203), upp_factor=np.float32(0.27294922)]

correderadiego, 2025-04-13:
I have been making some tests and I think that I found my issue. My grip is giving me more than the 100 degrees of freedom that the grip needs. So If I try to make the calibration with it it fails.

The solution will be modifying this limit or just adjust the hardware limit of the grip.

DipFlip, 2025-04-21:
I'm getting the same error when running the calibration
```bash
ValueError: No integer found between bounds [low_factor=np.float32(0.018066406), upp_factor=np.float32(0.018066406)]
```
Not sure I understand what you describe @correderadiego . Did you figure it out?

correderadiego, 2025-04-23:
My problem was that the grip gaves a wider angle that the software limits.
So I modify the feedtech class with my new limits. I increase the limits
step by step útil the error desapears.