TheRobotStudio/SO-ARM100 #63: Cannot insert my servo motor to the base printed in Bambu Lab P1S

ywiyogo, 2025-04-19:
@philfung ,
I've already printed the parts using the file Print_Leader_Bambu_P1P.3mf with Bambu Lab P1S. However, I cannot insert the servo motor to the base. It's just around 0.5 mm too tight.

[image]

I tested the gauge zero from stl_files_for_3dprinting/Gauges/Gauge_0.STL and it fits well.

The printer setting is as recommended in the README: 
* Material: PLA
* Nozzle Diameter and Precision: 0.4mm nozzle diameter at 0.2mm layer height.

What did I do wrong and which setting should I check in my printer?

Thanks!

philfung, 2025-04-19:
@ywiyogo hm, thanks for the report.  I've used this file to print 8 different arms, and the servo fits correctly when i use it (though I sometimes have to push it in with a lot of force).  Does the servo not fit at all, even with force?   If it doesn't fit at all, then I can try to print again and check.

ywiyogo, 2025-04-20:
Hi @philfung , yes even with a huge force, it got stuck and leaves about 0.5 cm space to the end. Then, I need to use a tool and huge force to pull my servo motor from the base. I thought your file will be fit for P1S too. I'll do some experiment to find out the cause of this issue.
I tried to mount the second part without the motor and it doesn't fit too.

[image]

philfung, 2025-04-20:
I'm sorry for this.  Ill try to print out a part in the next few days and see what's wrong

ywiyogo, 2025-04-20:
I think, I see the issue.
The servo motor has 24.72 mm witdh and 35 mm height on the back, like the following dimension from the specification:

[image]

My measurement in the base model shows 24.70 mm for the width and 32.90 mm height. The different in height is 2 mm and 0.02 mm in the width.

[image]

[image]

In another parts I see smaller dimension than the real dimension of the servo motor.

I check the diff in the history, and it shows that all the measurements in the base model are from the beginning the same. Thus, I wonder how many people don't have the issue during the assembly of this robot.

philfung, 2025-04-20:
[image]

Hi @ywiyogo , I just did a fresh print using the checked in pmf file on a Bambu P1P with PLA.  It fits fine with both a Waveshare 3215 (the servo you're using) and the Feetech 3215 (the recommended servo).  In addition I've known several others who've printed out using this file.

jbcayrou, 2025-04-21:
Hi!

I’m having the same issues as @ywiyogo . I had some trouble fitting the base with the motor 1 blocker (Base_08q-5.STL). The most difficult part for me was the Wrist_Roll_Pitch_08i-1.STL. I'm using a Creality K1C printer with PLA.

I managed to assemble the parts by sanding down the small corners where they were rubbing, but the fit is still very tight and requires a lot of force.
Is there a specific reason why the fit is so tight (the sides seem slightly smaller than the servo)?

Do you have additional settings on your printer?

philfung, 2025-04-23:
@jbcayrou hm, the pmf has all the settings (at least form Bambu P1P).  
@jbcayrou @ywiyogo One "hacky" solution is to edit the file slightly in a CAD program to make the opening bigger for your printers (For your convenience, here is a [CAD](https://www.tinkercad.com/things/ldsY0DOvMv6/edit?returnTo=%2Fdashboard%2Fdesigns%2F3d&sharecode=GvaS-DJQgCtVegXi9IygiKomz69APdIggxPPMUXBmtM&sharecode=GvaS-DJQgCtVegXi9IygiKomz69APdIggxPPMUXBmtM) file with the base, wrist and feetech servo which you can edit).

ywiyogo, 2025-04-24:
@jbcayrou thanks for sharing. As, I've been pointed out above that the opening in the model doesn't fit to the dimension of the servo motor. The 2 screenshots above is from BambuStudio and in FreeCAD is the same.
@philfung yes, I'm working to modify the opening for the servo motor in FreeCAD. Thanks for the CAD file.

dmosher42, 2025-05-09:
@philfung @ywiyogo 
Before editing the CAD file, you may want to try changing the [XY hole/contour compensation settings](https://wiki.bambulab.com/en/software/bambu-studio/xy-hole-contour-compensation) in Bambu Studio.

neta3200, 2025-12-27:
Can someone give a modifed STL for BambuLab P1S ?
or the dimensions he configured in the tinkercad stl that [philfung](https://github.com/philfung) sent here ?
[ywiyogo](https://github.com/ywiyogo) you fix that ? you can add your stl ?
thanks a lot .