TheRobotStudio/SO-ARM100 #68: Overhead Camera Mount unavailable for Europe

Yanis-Bouchelaghem, 2025-04-26:
Hello!

Thank you all for this amazing repo, and thank you @cmcgartoll for providing the [overhead camera mount design](https://github.com/TheRobotStudio/SO-ARM100/blob/main/Optional/Overhead_Cam_Mount/README.md).

However, the link for the [recommended camera](https://www.amazon.com/dp/B082X91MPP) is not available for europe. I am having a hard time finding the same camera model from the same manufacturer "XPCAM" in europe.

Could anyone please recommend an alternative camera that is compatible with the overhead camera mount design ?

Could I also suggest maybe having an alternative mount design that works with [camera modules](https://www.amazon.com/innomaker-Computer-Raspberry-Support-Windows/dp/B0CNCSFQC1/132-7372155-9780230?pd_rd_w=eYz4L&content-id=amzn1.sym.4c8c52db-06f8-4e42-8e56-912796f2ea6c&pf_rd_p=4c8c52db-06f8-4e42-8e56-912796f2ea6c&pf_rd_r=XC3EXZRSSXKDB1G0Z5D7&pd_rd_wg=1wTpn&pd_rd_r=932b1976-9ac7-4cef-9774-f0f9c3acb804&pd_rd_i=B0CNCSFQC1&psc=1) used for the [wrist cameras](https://github.com/TheRobotStudio/SO-ARM100/tree/main/Optional/Wrist_Cam_Mount_32x32_UVC_Module) ?

Thank you.

GuillaumeVW, 2025-04-29:
I'm planning on modifying the existing overhead mount design to work with the [Vinmooog Webcam](https://www.amazon.com.be/gp/product/B0BG1YJWFN).
This is the same camera used in [one of the optional wrist cam designs](https://github.com/TheRobotStudio/SO-ARM100/tree/main/Optional/Wrist_Cam_Mount_Vinmooog_Webcam).
Once I have a working modification for the overhead mount, I can share the updated files if there's interest.

philfung, 2025-04-29:
> maybe having an alternative mount design that works with camera modules

++ agree that an overhead mount that allows a plug-in 32x32 camera module would be most scalable + universal
@cmcgartoll what do you think ?

philfung, 2025-04-29:
@Yanis-Bouchelaghem as another alternative, i'm 80% sure this [cam](https://fr.aliexpress.com/item/1005006949399519.html) will yield similar results to the US cam, as they are similar in dimension

cmcgartoll, 2025-04-29:
@philfung, the 32x32 would be a nice addition to the overhead camera although FOV may be weird. I can work on that. 

I will also just open source the STEP/Inventor files so anyone can build on it. @GuillaumeVW that might be helpful to you so you can more easily build on top of it!

Yanis-Bouchelaghem, 2025-05-01:
I have just ordered the [alternative camera](https://fr.aliexpress.com/item/1005006949399519.html) you suggested @philfung, I will let you know if it is compatible once I receive it.

Thank you for the suggestion! I am also looking forward to the alternative designs for the overhead mount.

philfung, 2025-05-06:
[image]

[image]

Yanis-Bouchelaghem, 2025-05-22:
Hello,

> [@Yanis-Bouchelaghem](https://github.com/Yanis-Bouchelaghem) as another alternative, i'm 80% sure this [cam](https://fr.aliexpress.com/item/1005006949399519.html) will yield similar results to the US cam, as they are similar in dimension

Unfortunately, the camera suggested by @philfung fits the overhead mount, but has an extremely small POV, I can confirm that this specific camera is inadequate.

The new [32x32 UVC camera module design](https://github.com/TheRobotStudio/SO-ARM100/pull/88) looks much better suited. I will print it and try it soon.

Thank you for the contribution !

I will now close this issue.

closed 2025-05-22 by Yanis-Bouchelaghem

philfung, 2025-05-22:
@Yanis-Bouchelaghem my deep apologies that this suggestion didn't work out.  Have you tried increasing the resolution of the camera (1280 x 800)?  This will increase FOV.

https://github.com/huggingface/lerobot/blob/main/lerobot%2Fcommon%2Frobot_devices%2Frobots%2Fconfigs.py#L256

"Downside" is the camera training files will be slightly bigger.