TheRobotStudio/SO-ARM100 #71: SO-100 components(without 3D Parts) can be used for So-101?

Huangruopeng, 2025-04-29:
I just bought SO-ARM100 Servo Motor Kit (Without 3D Printed Parts).
Can I equip it with New SO-101 3D printed parts?

pkooij, 2025-04-29:
Yes you can! From the motors perspective only the leader arm motors for SO-101 are differently geared

acyanbird, 2025-05-03:
@pkooij Hi that might be a silly question. I have a pair of ( follower and leader ) arm, and I want to update to so101 version. But how can I modify the gear ratio base on the so100 leader arm's servo?

matbee-eth, 2025-05-05:
> [@pkooij](https://github.com/pkooij) Hi that might be a silly question. I have a pair of ( follower and leader ) arm, and I want to update to so101 version. But how can I modify the gear ratio base on the so100 leader arm's servo?

you'd need the replacement gears- or purchase the kit from alibaba

mazytomo, 2025-05-06:
Hi, let me see if I am understanding correctly.

I have twelve 12v motors for SO100, which means I have to buy six 7.4v motors for the leader arm, right?
 > the leader arm is always 7.4v for the SO101.

Huangruopeng, 2025-05-15:
> > [@pkooij](https://github.com/pkooij) Hi that might be a silly question. I have a pair of ( follower and leader ) arm, and I want to update to so101 version. But how can I modify the gear ratio base on the so100 leader arm's servo?
> 
> you'd need the replacement gears- or purchase the servo kit from alibaba

So the leader servo is different? Can you list servo brand and version? Thank you so much? or how to replace gears?

dannyyy, 2025-09-13:
I also have the Motor Kit with 12x 7.4V (C001). For the follower arm, no change is required.

What happens if I use the same gear for all motors in the leader arm? Do I need to adjust something in the software (servo register), calibration, ...?

Or is it impossible to use these motors for the leader arm?