TheRobotStudio/SO-ARM100 #75: Manual Leader Calibration

matbee-eth, 2025-05-05:
Is it possible to simply rotate each servo to its maximum CW/CCW rotation and use that for calibration instead of/along-side the 3-pose calibration?

I believe I rotated some of my servos before assembly so some of them act up.

pkooij, 2025-05-09:
Yes this is something we are working on and will be released in the coming weeks

matbee-eth, 2025-05-09:
Until then, is there a way to monitor a motors current position and set it as its "zero" or whatever formula the calibration scripts output?

pkooij, 2025-05-22:
Hi @matbee-eth If you now recalibrate both arms with LeRobot it should be fixed, we added an additional calibration step (first step) where you need to place the robot in center position for each joint.