TheRobotStudio/SO-ARM100 #119: [low priority] Mount Helper part is not oriented optimally for printing

cHemingway, 2025-07-12:
Thanks for this project, its been great fun working through the prints, and there is lots of guidance for most parts.

However, I noticed the [Mount Helper](Optional/Mount_Helper/stl/mount_helper v2.stl) part is designed with teardrop holes, but these do not match its default Z orientation, nor are they the right way up for fastest printing (the largest flat side should be on the buildplate I think).

If we could turn the hole around, it wouldn't need supports. TBH I'm not sure if the hole is even needed? I don't see it being used in the video.

Tagging @fred3105 as he uploaded the file. Do you have the original CAD? I would be very happy to contribute an updated version if you can provide it, even the STEP file would do. 

Print times on my Prusa Mini+ with a 0.6mm nozzle, 0.4mm layer height as follows:

Here is the original orientation with supports, **2.5 hours**:

[image]

Here is it in an optimal orientation, with no supports, **1.15 hours**

[image]

fred3105, 2025-09-25:
Hey @cHemingway, 

I sent the step file to the email address on your GitHub profile
As @dmosher42 mentions, I simply added a plastic block to the small elbow piece, nothing fancy, 

You're more than welcome to improve it

cHemingway, 2025-09-29:
Thanks for the CAD!

Remodelling it to remove the holes (I kept all other dimensions), it actually only saves 15 minutes (1h31 to 1h15m ~20%) with organic supports. Less of a saving than I expected

[image]

[image]

However, if I make the base a bit thinner (6mm instead of 10mm) this saves another 15 minutes, so its down to 1 hour from 1.5 hours. I don't know why the base is quite that thick, seems like thinner might work.

[image]

I can put the new CAD in a pull request if you like? I will print it out first and test it if so.